nebrius/raspi-board #2: Raspberry Pi 2 Compatibility

braytron, 2015-02-17:
I just bought a pi 2 and started looking at doing some node programming to control the gpio pins. I am very new to all of this but wanted to let you know what i'm seeing and offer my pi 2 up for any testing you need done. I would love to help. I've got node installed on my pi2 and in the node repl I ran the following commands to come up with 10489921 as the board revision based on your index.js.

<code>var procInfo = fs.readFileSync('/proc/cpuinfo').toString();</code>

<code>var rev = procInfo.match(/Revision\s*:\s*(.*)/);</code>

<code>rev = parseInt(rev && rev[1] || '0', 16);</code>

I made some modifications on my index.js as follows:

var BOARD_REVISIONS = {
  0x00: 'Unknown',
  0x02: 'B1',
  0x03: 'B1',
  0x04: 'B2',
  0x05: 'B2',
  0x06: 'B2',
  0x07: 'A',
  0x08: 'A',
  0x09: 'A',
  0x0D: 'B2',
  0x0E: 'B2',
  0x0F: 'B2',
  0x10: 'BPLUS',
  0x12: 'APLUS',
  10489921: 'PI2'

And here:

switch (BOARD_REVISIONS[rev]) {
  case 'A':
    throw new Error('Rev A boards are not yet supported.');
    break;
  case 'B1':
    pins = B1;
    break;
  case 'B2':
    pins = B2;
    break;
  case 'APLUS':
  case 'BPLUS':
  case 'PI2':
    pins = BPLUS;
    break;
  default:
    throw new Error('Unknown board revision ' + revision);
}

Not sure if this helps but i would be willing to follow any direction you give. Thanks for your contributions to the node and raspberry pi communities.

bryan-m-hughes, 2015-02-17:
Funny you mention this, I was working on this code this morning :). I have a solution coded up that works on my RPi2, I just need to test backwards compatibility to make sure the other boards still work.

I'm taking advantage of the opportunity to clean up how the detection works, and how the board revisions are represented. I'm thinking now that I'm going to call everything ```RPi[1|2]_[model]```, e.g. ```RPi2_B``` and ```RPi1_B2```.

From the research I've done, the ID you listed is only one of (at least) two possible revision IDs for the RPi 2 Model B. The revision is now also based on where it was manufactured and encodes how much memory it has, not just which model number it is. Boards seem to be either manufactured in the UK or in China from what I can tell. So fun times.

bryan-m-hughes, 2015-02-18:
Added support with v2.0.0 of raspi-board and v2.0.7 of raspi-io

closed 2015-02-18 by bryan-m-hughes