nebrius/raspi-board #3: Revision lookup does not handle boards which have overclocked

shortbloke, 2015-07-10:
The revision number reported by cpuinfo on my is 100000e not just the 000e it's that is expected. Accoridng to http://elinux.org/RPi_HardwareHistory a 1000 prefix indicates that the board has been overclocked at some point in time and therefore voiding the warranty. Once the bit is set, it can't be cleared. 
As a simple workaround I just added by revision to the list, but it would be good if you could handle the general case where the revision numbers may be prefixed by 1000. 

Value from: /proc/cpuinfo:
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 2.00
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2708
Revision        : 100000e
Serial          : 00000000e2ae5fb3

nebrius, 2015-07-10:
Thanks for letting me know @shortbloke! I actually haven't overclocked an RPi myself, so I had no idea it did this.

By chance, do you know if "1000" is also appended to RPi 2 overclocked boards? Their identifier scheme is different than the RPi 1, so it may behave differently and the e-linux article doesn't mention what happens.

If not, no worries, I can always overclock an RPi myself and find out :) It'll just be a while before I have a chance to do so.

shortbloke, 2015-07-10:
I don't have a Pi2 (yet, just ordered one) and before I tripped over this bug I wasn't aware they set this warranty bit if overclocked or how it was exposed ;-)

Anyway a bit of searching I stumbled over: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=98367&start=250 

Where someone overclocked their Pi2 and the revision reported was "1a21041", so gained a 1 prefix to the usual pi2 model number.

Maybe the solution is to ignore the 1 and stip the leading zero's from the revisions. That might be sufficient to handle Pi and Pi2 use cases in a single expression?

Hopefully that's enough to come up with a change and avoid the need for you to overclock your pi.
Martin

nebrius, 2015-07-14:
Fixed with v3.3.3. Thanks again for the bug report!

closed 2015-07-14 by nebrius