nebrius/raspi-board #4: Pi Zero compatibility

ekryski, 2015-12-04:
It's obviously not supported as I ran into this error:

```
/home/pi/led/node_modules/raspi-io/node_modules/raspi-board/lib/index.js:374
    throw new Error("Unknown board revision " + rev);
    ^

Error: Unknown board revision 900092
    at Object.<anonymous> (/home/pi/led/node_modules/raspi-io/node_modules/raspi-board/lib/index.js:374:11)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Object.<anonymous> (/home/pi/led/node_modules/raspi-io/lib/index.js:48:19)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
```

Totally willing to do a PR if you can point me in the right direction. Going to start spelunking the code right now.

ekryski, 2015-12-05:
I have a PR open for this (https://github.com/nebrius/raspi-board/pull/6) and have verified that it works. Would love to get this merged so that I can install from the official repo.

nebrius, 2015-12-05:
Thanks for the PR! I've been looking at it, and I think the code looks OK. I'm just trying to figure out a good way to test it since my Pi Zero is still on backorder.

ekryski, 2015-12-05:
might be a while then :disappointed:. Maybe I could send a video? (only half kidding)

nebrius, 2015-12-09:
OK, v2.3.0 of raspi-board has been published! The update hasn't been propagated to the other dependencies yet, but that should happen soon!

nebrius, 2015-12-09:
all packages updated, RPi Zero support is in!

closed 2015-12-09 by nebrius

ekryski, 2015-12-09:
Woot! Thanks :smile: