nebrius/raspi-board #5: Add tests per board type

ekryski, 2015-12-04:
I quickly poked at the tests and they assume the latest pi 2 board. It would be awesome to have tests against each board revision.

I'd also be down to help on that front.

nebrius, 2015-12-05:
Can you elaborate on what we would gain by simulating other boards? I'm not really sure we would gain anything, but I may be missing something.

The board is simulated in the test, since the tests are the same across board revisions, plus I don't ever actually run them on an actual RPi (I run them in OS X).

ekryski, 2015-12-05:
I hadn't realized that they were simulated until after I wrote this. Do you think there is any value in testing against the actual boards?

My thought was maybe writing tests to make sure that the other pinout definitions are correct based on at least different simulated board types. Maybe to prevent future breakage?

Maybe this is just writing tests for tests sake then... totally your call. I'd write them but it may not be worth while.

nebrius, 2015-12-05:
I think there is value in writing tests like that, however I don't think it's possible to write them at this module's level.

The current tests don't actually confirm that the pinout is correct. They basically assume that the pin mapping is correct, and really just test some funky edge cases.

Getting a little more existential/philosophical for a moment, consider the questions: "how do we test to make sure the pinout definition is correct"? I don't think there's an answer, myself.

I think the answer to this is more holistic/bigger picture testing at the raspi-io level using the architecture I laid out in https://github.com/nodejs/hardware/issues/17

ekryski, 2015-12-05:
Thanks for linking that discussion. I think having a standardized test system for hardware is a great idea! After reading that thread I totally understand what you mean.

So I agree that it's not possible to determine whether the pins _function_ correctly at this level but I do think there is value in writing tests for our pin mappings, at least to guard against any future breakage.

In reality, these pin mappings aren't really assumed because we do have pinout specs for the boards. The assumption is more that they work as expected, which definitely needs to be tested in fashion that you described in https://github.com/nodejs/hardware/issues/17.

So... the tests I would write would be based on a given board rev, I expect to have these pins available, maybe at least until we have tests at a lower level that verify I can send PWM or I2C through the expected pin. Thoughts?

nebrius, 2015-12-06:
We do have the pinout specs, yes, but they have to be manually copied and converted into the code and tests. Basically what we would end up with is tests designed to check if we manually copied the spec over properly in the code...by manually copying the spec over into the tests. I don't see any real value in that myself.

Or am I missing something?

ekryski, 2015-12-06:
Nah fair enough. It's pretty negligible value then except to catch if someone ever accidentally changes them. Feel free to close this out.

nebrius, 2015-12-06:
Thanks, I'll close then. If you git blame, you'll see that the last time I modified the mappings was in 2014, so I'm not worried about them getting changed :D

closed 2015-12-06 by nebrius