nebrius/raspi-board #11: RPi Zero: Unknown board revision 920093

goliatone, 2016-10-12:
Using Raspberry Pi Zero 1.3. Getting the following error:

```
at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
/usr/src/app/node_modules/raspi-board/lib/index.js:380
    throw new Error('Unknown board revision ' + rev);
    ^

Error: Unknown board revision 920093
```

goliatone, 2016-10-17:
Any chances this would get committed?

nebrius, 2016-10-18:
raspi-board has been updated and published already as version 4.0.0, but it still needs to be updated in the other dependencies. I'm traveling for conferences right now, and don't have the ability to do regression testing before publishing, so I'm wary of bumping them at this exact moment.

goliatone, 2016-10-18:
ok, makes sense.

nebrius, 2016-11-01:
Tested, and merged in to raspi-io 6.1.1

closed 2016-11-01 by nebrius