nebrius/raspi-board #14: Revision b03111, still throws error

irakandjii, 2019-07-16:
Hi I reported bo3111 earlier as the Raspi Pi 4, 2 gigabyte unit.  It was confirmed that the update to this code was made.  Yet on my system it still throws the error.  I have a fresh install of "buster" full package on a new raspi 4, with a 128gB microSD card.
I have done update/upgrade cycle and reinstalled johnny-five and raspi-io.  Am I doing something wrong?
Also the LED does not flash, when I exit with ctrl-c I get "sigHandler: Unhandled signal 2, terminating"  I do not know if this is normal.  
I have followed the installation instructions to the letter, when I use npm i -g johnny-five raspi-io I get a lot of warnings.  I do not know if this is normal.

nebrius, 2019-07-16:
Hi @irakandjii. There's a couple of things to separate out, as multiple things are happening at the same time here that are most likely unrelated

1) The issue you reported the other day was about a warning, not an error. The fix was to prevent the warning from being printed, but no behavior change was needed as it already worked.
2) The LED does not flash
3) You get "sigHandler: Unhandled signal 2"
4) You are using the `-g` flag with `npm install`. This is not supported in raspi-io or any library that is `require()`'d in Node.js. General rule of thumb is to never use the `-g` flag unless the instructions explicitly say to use it.
5) You get warnings on install. These are most likely build warnings from [serialport](https://github.com/serialport/node-serialport), which is expected and can be ignored.

Going through these, 1) is already fixed, 4) is user error, and 5) can be ignored. So let's talk about the others.

**Problem 2)**

This is most likely a wiring issue. Have you double checked that your LED is plugged in the right direction? If it's backwards, nothing will happen. It doesn't hurt an LED to be plugged in backwards, but it doesn't work properly.

It could also be with the resistor you're using in series with the LED. What is the value of this resistor? (If you're not using one, you should stop ASAP and add one, as this can damage your Raspberry Pi in some cases).

It could also be the case that you don't have a proper ground connection. Is your ground connection from your LED connected to ground on the Raspberry Pi?

Lastly, it could be that you have the LED plugged in to the wrong pin. Which pin alias are you using in code, and which pin did you connect the LED to?

**Problem 3)**

Honestly I have no idea what this one could be. I don't think it'll hurt anything though. I'll see if I can reproduce it, but you can _probably_ ignore this one.

irakandjii, 2019-07-16:
Thank you for the detailed response.

1) I still get the warning so I must be doing something wrong.

2) The LED works with both scratch2 and Python so I know it works, the wiring is ok.  There is something else going on.

4) I will try another fresh install and see if I can get it working, I am sorry to be a bother but I am so new to johnny-five that it is difficult  to know what to ask much less fix it.

5) Yes, the warnings seem to all come from serialport which does work, I have a test program confirming 2 way serial comms.

nebrius, 2019-07-16:
> I still get the warning so I must be doing something wrong.

Oh I think I know what it is, you might still be on the older version because of how package-lock.json works. Can you run `npm update` and try again?

> The LED works with both scratch2 and Python so I know it works, the wiring is ok.  There is something else going on.

I almost forgot, it could also be how you're referring to the pin in code. Have you read through https://github.com/nebrius/raspi-io/wiki/Pin-Information? tl;dr, if you're passing in a number instead of a string, it's most likely not doing what you think it is, and would explain why the LED isn't blinking. I never use numbers myself, because they're abstract and don't map to, well, anything (e.g. GPIO18 is `1`, not `18`)

> I will try another fresh install and see if I can get it working, I am sorry to be a bother but I am so new to johnny-five that it is difficult  to know what to ask much less fix it.

No bother at all :)

> Yes, the warnings seem to all come from serialport which does work, I have a test program confirming 2 way serial comms.

You can definitely ignore these warnings then

irakandjii, 2019-07-16:
I installed a new SD card with the most recent duster on it, then installed exactly as recommended .   So all the software is “fresh”

For the test I used (cut and paste) the source code on https://github.com/nebrius/raspi-io website.   I learned I had to add the .RaspiIO to the require statement.  Most tutorials and books leave this out.

Even after the npm update. I still have the same problems I get the warning; and no LED activity,

I use the string ‘P1-7’  I also tried ‘GPIO4’,  I confirmed I am on GPIO4 and retested with scratch2 and I can light the LED within that environment.

Weird,  I agreed this must usually work, because multiple books and tutorials  take the same approach.  Frustrating I know…

nebrius, 2019-07-16:
> For the test I used (cut and paste) the source code on https://github.com/nebrius/raspi-io website.   I learned I had to add the .RaspiIO to the require statement.  Most tutorials and books leave this out.

Yeah, the `.RaspiIO` that's a recent change in Raspi IO 10.x.x, and almost all documentation was written before Raspi IO 10 was released. Such is the nature of third party documentation I have no control over :-/

Can you provide the output of `npm ls raspi-board`?

Also, can you take a photo of your wiring and paste it here?

irakandjii, 2019-07-16:
This is the output, I am sending 1 at a time

irakandjii, 2019-07-16:
this is the wiring

irakandjii, 2019-07-16:
Hopefully the pictures got through the first is the output the second is the wiring

nebrius, 2019-07-16:
The pictures didn't make it through unfortunately. You may need to post them directly on GitHub's site, instead of through email.

irakandjii, 2019-07-16:
[image]
This is the output, you can see the error / warning messages too.

[image]
this is the wiring
[image]
This is the code to light the LED in scratch2 using GPIO4 (sorry upside down)
[image]
this is the LED lit with scratch2

irakandjii, 2019-07-16:
I did not mean to close this

nebrius, 2019-07-16:
Thanks. That all does look correct to me. My RPi 4 _just_ arrived, so I'll see if I can reproduce this as soon as I get a chance. It _might_ be next week though.

If you feel like hacking on this on your own in the mean time, try dropping in some `console.log` statements or use a debugger to see what's going on in these lines of code: https://github.com/nebrius/raspi-board/blob/master/dist/index.js#L793-L839. This does the board detection, and _should_ be set to one of these constants: https://github.com/nebrius/raspi-board/blob/master/dist/index.js#L75-L76

irakandjii, 2019-07-16:
Np, take care and thank you…   good hunting…

jackcannon, 2019-07-17:
I've just seen this after having originally raised a similar/same issue on the Raspi-IO repo.
From a quick look at the changes, it seems that b03111 & c03111 were added as board revisions as VERSION_4_MODEL_B, but that wasn't used anywhere?
Would it not need to be added to the switch at https://github.com/nebrius/raspi-board/blob/master/src/index.ts#L839 ?

irakandjii, 2019-07-18:
Jack are you able to blink the LED?

nebrius, 2019-07-18:
D'oh, thank you @jackcannon! That's exactly the issue. I just published a new version of raspi-board with the fix, which you should get with `npm update`

irakandjii, 2019-07-18:
On my older Pi 3, everything works.  Just my Pi 4 is not.  The message about the board is now supressed.
The Pi 4 does not return Available, Connected, Repl Initialized  ...  like the Pi 3 does.

jackcannon, 2019-07-18:
@irakandjii I haven't tried in the last week or so, but will try to with the latest version tonight, and keep you posted.
@nebrius Thanks for that! Would've opened a PR, but it was nearly midnight when I saw it!

irakandjii, 2019-07-18:
@jackcannon ok, ty!  
Doing some tests now, but I think I am in over my head.  I cannot find why the Pi4 hangs.

jackcannon, 2019-07-18:
Can confirm that it hangs when trying to create the Board.

Raspi-io example code, with added logs:
```javascript
console.log('a');

const Raspi = require('raspi-io').RaspiIO;

console.log('b');

const five = require('johnny-five');

console.log('c');

const board = new five.Board({
  io: new Raspi()
});

console.log('d');

board.on('ready', () => {
  console.log('e ready');

  // Create an Led on pin 7 (GPIO4) on P1 and strobe it on/off
  // Optionally set the speed; defaults to 100ms
  (new five.Led('P1-7')).strobe();
});
```

Output:
```
pi@raspberrypi:~/Projects/raspi-io-tests $ sudo node test1.js
a
b
c

```

jackcannon, 2019-07-18:
Correction: it hangs when creating the Raspi io object

jackcannon, 2019-07-18:
I've had a poke around, but not found any fixes.
Added a load of logs to the RaspiIO code, and and it looks like it's hanging when creating the J5IO, but when comparing Pi3 and Pi4, can't find any differences in the options provided.

The only thing I found, which doesn't fix the problem, but may be a step in the right direction, is that VERSION-4 wasn't in the enableSerial condition https://github.com/nebrius/raspi-io/blob/master/src/index.ts#L71

Pi3 Options:
```
{
  pluginName: 'Raspi IO',
  pinInfo: {
    '0': { pins: [Array], peripherals: [Array], gpio: 17 },
    '1': { pins: [Array], peripherals: [Array], gpio: 18 },
    '2': { pins: [Array], peripherals: [Array], gpio: 27 },
    '3': { pins: [Array], peripherals: [Array], gpio: 22 },
    '4': { pins: [Array], peripherals: [Array], gpio: 23 },
    '5': { pins: [Array], peripherals: [Array], gpio: 24 },
    '6': { pins: [Array], peripherals: [Array], gpio: 25 },
    '7': { pins: [Array], peripherals: [Array], gpio: 4 },
    '8': { pins: [Array], peripherals: [Array], gpio: 2 },
    '9': { pins: [Array], peripherals: [Array], gpio: 3 },
    '10': { pins: [Array], peripherals: [Array], gpio: 8 },
    '11': { pins: [Array], peripherals: [Array], gpio: 7 },
    '12': { pins: [Array], peripherals: [Array], gpio: 10 },
    '13': { pins: [Array], peripherals: [Array], gpio: 9 },
    '14': { pins: [Array], peripherals: [Array], gpio: 11 },
    '15': { pins: [Array], peripherals: [Array], gpio: 14 },
    '16': { pins: [Array], peripherals: [Array], gpio: 15 },
    '21': { pins: [Array], peripherals: [Array], gpio: 5 },
    '22': { pins: [Array], peripherals: [Array], gpio: 6 },
    '23': { pins: [Array], peripherals: [Array], gpio: 13 },
    '24': { pins: [Array], peripherals: [Array], gpio: 19 },
    '25': { pins: [Array], peripherals: [Array], gpio: 26 },
    '26': { pins: [Array], peripherals: [Array], gpio: 12 },
    '27': { pins: [Array], peripherals: [Array], gpio: 16 },
    '28': { pins: [Array], peripherals: [Array], gpio: 20 },
    '29': { pins: [Array], peripherals: [Array], gpio: 21 }
  },
  platform: {
    base: {
      init: [Function: init],
      getActivePeripherals: [Function: getActivePeripherals],
      getActivePeripheral: [Function: getActivePeripheral],
      setActivePeripheral: [Function: setActivePeripheral],
      getPinNumber: [Function: getPinNumber]
    },
    gpio: {
      PULL_DOWN: 1,
      PULL_UP: 2,
      PULL_NONE: 0,
      createDigitalInput: [Function: createDigitalInput],
      createDigitalOutput: [Function: createDigitalOutput]
    },
    led: { createLED: [Function: createLED] },
    pwm: { createPWM: [Function: createPWM] },
    i2c: { createI2C: [Function: createI2C] }
  },
  i2cIds: { DEFAULT: 1 }
}
```

Pi4 Options:
```
{
  pluginName: 'Raspi IO',
  pinInfo: {
    '0': { pins: [Array], peripherals: [Array], gpio: 17 },
    '1': { pins: [Array], peripherals: [Array], gpio: 18 },
    '2': { pins: [Array], peripherals: [Array], gpio: 27 },
    '3': { pins: [Array], peripherals: [Array], gpio: 22 },
    '4': { pins: [Array], peripherals: [Array], gpio: 23 },
    '5': { pins: [Array], peripherals: [Array], gpio: 24 },
    '6': { pins: [Array], peripherals: [Array], gpio: 25 },
    '7': { pins: [Array], peripherals: [Array], gpio: 4 },
    '8': { pins: [Array], peripherals: [Array], gpio: 2 },
    '9': { pins: [Array], peripherals: [Array], gpio: 3 },
    '10': { pins: [Array], peripherals: [Array], gpio: 8 },
    '11': { pins: [Array], peripherals: [Array], gpio: 7 },
    '12': { pins: [Array], peripherals: [Array], gpio: 10 },
    '13': { pins: [Array], peripherals: [Array], gpio: 9 },
    '14': { pins: [Array], peripherals: [Array], gpio: 11 },
    '15': { pins: [Array], peripherals: [Array], gpio: 14 },
    '16': { pins: [Array], peripherals: [Array], gpio: 15 },
    '21': { pins: [Array], peripherals: [Array], gpio: 5 },
    '22': { pins: [Array], peripherals: [Array], gpio: 6 },
    '23': { pins: [Array], peripherals: [Array], gpio: 13 },
    '24': { pins: [Array], peripherals: [Array], gpio: 19 },
    '25': { pins: [Array], peripherals: [Array], gpio: 26 },
    '26': { pins: [Array], peripherals: [Array], gpio: 12 },
    '27': { pins: [Array], peripherals: [Array], gpio: 16 },
    '28': { pins: [Array], peripherals: [Array], gpio: 20 },
    '29': { pins: [Array], peripherals: [Array], gpio: 21 }
  },
  platform: {
    base: {
      init: [Function: init],
      getActivePeripherals: [Function: getActivePeripherals],
      getActivePeripheral: [Function: getActivePeripheral],
      setActivePeripheral: [Function: setActivePeripheral],
      getPinNumber: [Function: getPinNumber]
    },
    gpio: {
      PULL_DOWN: 1,
      PULL_UP: 2,
      PULL_NONE: 0,
      createDigitalInput: [Function: createDigitalInput],
      createDigitalOutput: [Function: createDigitalOutput]
    },
    led: { createLED: [Function: createLED] },
    pwm: { createPWM: [Function: createPWM] },
    i2c: { createI2C: [Function: createI2C] }
  },
  i2cIds: { DEFAULT: 1 }
}
```

nebrius, 2019-07-22:
I finally managed to test the code as well, and I can confirm that the problem is that `ready` event is not being fired. On top of that, there is some Johnny-Five code that prints some information to the command line _prior_ to the `ready` event being fired that is not being printed.

Right now, my hunch is that Johnny-Five is hanging somewhere for some reason. I'm going to dive into this today and figure out what's going on, and which module is responsible (I'm about 95% certain it's not in raspi-board or raspi-serial)

nebrius, 2019-07-22:
> The only thing I found, which doesn't fix the problem, but may be a step in the right direction, is that VERSION-4 wasn't in the enableSerial condition https://github.com/nebrius/raspi-io/blob/master/src/index.ts#L71

I do need to fix this logic, but it won't be causing this issue.

irakandjii, 2019-07-22:
Thank you for your efforts on this

nebrius, 2019-07-23:
Follow up: I managed to isolate the issue, and it's in one of my underlying dependencies. I filed an issue at https://github.com/fivdi/pigpio/issues/79.

nebrius, 2019-08-07:
In case you didn't see it in the linked issue, https://github.com/fivdi/pigpio/issues/79#issuecomment-515209051 describes the issue and how to work around it.