nebrius/raspi-board #18: Unknown board revision 2a02082 - Raspberry Pi 3 B

mikethorpe, 2020-06-19:
Hi,

On my 3 b I'm getting:

`Unknown board revision 2a02082, assuming Raspberry Pi Zero/2/3 pinout. Unless you are running a compute module or very old RPi this should work fine. Please report this board revision in a GitHub issue at https://github.com/nebrius/raspi-board.
/home/pi/projects/BackEndApp/node_modules/bindings/bindings.js:121
        throw e;
        ^

Error: /home/pi/projects/BackEndApp/node_modules/@serialport/bindings/build/Release/bindings.node: wrong ELF class: ELFCLASS64
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:807:18)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at bindings (/home/pi/projects/BackEndApp/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/home/pi/projects/BackEndApp/node_modules/@serialport/bindings/lib/linux.js:2:36)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)`

Code:
`const five = require('johnny-five');
const Raspi = require('raspi-io').Raspi;
const board = new five.Board({
    io: new Raspi({ enableSerial: true }),
});

board.on("ready", function () {
    var led1 = new five.Led(29);
    led1.blink();
});`

Any help appreciated :)

mikethorpe, 2020-06-19:
Looks as though the error with serial port was due to it being built on the wrong platform. I fixed that. However, I'm still receiving the board revision error. On checking src/index.ts there is no board revision for '2a02082'.

nebrius, 2020-06-20:
The message you're receiving is just a warning, not an error. It shouldn't have any negative impacts on the operation of this module. Is anything not working for you?

mikethorpe, 2020-06-21:
I did some more digging and it still works with the warning. Apologies for the false alarm.

nebrius, 2020-06-22:
I've been thinking more and more lately that I'm going to remove that warning entirely. The pinout for the Raspberry Pi hasn't changed in many years, and I doubt they'll change it for future boards.

nebrius, 2020-07-16:
This revision has been added to v7.3.2 of raspi-board

closed 2020-07-16 by nebrius